codebutler/farebot #97: Exception while reading card

bacobart, 2015-02-12:
I'm getting an error "Attempt to invoke virtual method 'short android.os.Bundle.getShort(java.lang.String)' on a null object reference" while reading a card. The same card and phone used to work fine, but this error started to appear after the phone was upgraded to Android Lollipop. I've tried both the version in Google Play and have build the latest FareBot from git, but both have the same problem. I'm using a dutch OV-Chipkaart and HTC One M8, but I think the problem is related to Lollipop as this used to work before. Any ideas?

Stacktrace:
02-12 13:38:16.099  18556-18556/com.codebutler.farebot E/com.codebutler.farebot.activity.ReadingTagActivity﹕ Attempt to invoke virtual method 'short android.os.Bundle.getShort(java.lang.String)' on a null object reference
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ java.lang.NullPointerException: Attempt to invoke virtual method 'short android.os.Bundle.getShort(java.lang.String)' on a null object reference
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at android.nfc.tech.NfcA.<init>(NfcA.java:92)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at android.nfc.tech.NfcA.get(NfcA.java:78)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at android.nfc.tech.MifareClassic.<init>(MifareClassic.java:166)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at android.nfc.tech.MifareClassic.get(MifareClassic.java:156)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at com.codebutler.farebot.card.classic.ClassicCard.dumpTag(ClassicCard.java:70)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at com.codebutler.farebot.card.Card.dumpTag(Card.java:66)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at com.codebutler.farebot.activity.ReadingTagActivity$1.doInBackground(ReadingTagActivity.java:83)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at com.codebutler.farebot.activity.ReadingTagActivity$1.doInBackground(ReadingTagActivity.java:78)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at android.os.AsyncTask$2.call(AsyncTask.java:288)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at java.util.concurrent.FutureTask.run(FutureTask.java:237)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:231)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
02-12 13:38:16.099  18556-18556/com.codebutler.farebot W/System.err﹕ at java.lang.Thread.run(Thread.java:818)

codebutler, 2017-08-07:
Please download the latest version from Google Play and open a new ticket if you're still having trouble.

closed 2017-08-07 by codebutler

emveeha, 2017-08-09:
I still have the same issue while scanning an OV-chipkaart using the latest version from Google Play.