codebutler/farebot #100: Pasmo is treated like Suica?

shanen0, 2015-03-24:
The Pasmo card is not handled correctly, but is treated like a Suica card. Most importantly, the Suica card only holds 20 transactions in the history, while Pasmo has 100.
Secondary problem is that I can't figure out how to convert from XML to something more usable. Only the one export option?

yaiu, 2017-02-24:
Just bumping this issue with a confirmation that ICOCA and Kitaca cards are treated as Suica.

codebutler, 2017-02-25:
I haven't been able to find any data on the card that indicates the specific type. Please let me know if you have any information on this.

shanen0, 2017-06-11:
Not sure if this message is going to "codebutler", but I guess all I can say is that dumping the data shows different numbers of transactions recorded in the cards. (My experience is limited to Suica and Pasmo, and I don't know about the two kinds of cards yaiu mentioned.)

Actually I haven't even been using this project lately, but switched to something called 'Suica & Pasmo Reader' (in katakana). What I was really looking for was something more powerful than any of the programs I was able to find, but it appears that no one is doing anything along those lines. As usual, I think it comes back to motivations and therefore I look at the financial models... (I wish it were possible to pledge a donation towards the software I want and if enough people agreed, then our donations would fund it...) 

What I really wanted (until I gave up) was a program that would look at the card data from an analytic perspective to help me detect if there were any funny charges sneaking in there. Essentially that's a database that would help separate the normal repeated charges from the new and possibly funny ones. 

As a programmer, I was even hoping to find such projects on Github where I could do some of the grunt work in exchange for improving my Python skills, but...

codebutler, 2017-07-02:
@shanen0 Can you please link to the "Suica & Pasmo Reader" app you're using, and can you explain exactly what it supports that FareBot is missing?

JanakBca1992, 2019-03-12:
Is it possible to read more than 20 record from Suica card?

LunaticNeko, 2019-08-23:
What I know about these cards (from own experience) are:

* Japan Railways-issued cards (SUICA/ICOCA) and PASMO all seem to be "SUICA-compatible." There should be (but I don't know the details) unique identifiers that distinguish each issuer or something. My ICOCA's ID (physical number printed on the card) starts with JW, probably stands for "JR West." SUICAs begin with SU, IIRC. I do not have one with me.
* JR and Kintetsu ticket machines can pull only latest 20 entries from an ICOCA. I don't think any local tool can access deep history of a card if all you have is the card itself.

codebutler, 2026-02-14:
Fixed in KMP rewrite. Pasmo and Suica are distinguished via separate service code sets. Block reading uses unbounded loop — no 20-record cap.