codebutler/farebot #103: Support for IstanbulKart

jnuahs, 2015-05-14:
I'm new to this NFC programming, but I'm interested in helping get support for IstanbulKart (aka "Akbil") for the Istanbul, Turkey transit system.

I've scanned one of my cards and have lots of information, but some that I think is relevant:

IC Type: MIFARE DESFire EV1 (MF3ICD41)

Vendor ID: NXP
Protocol: ISO/IEC 14443-2 and -3

Taginfo provides a lot more information, but I'm not sure what's useful. I'm able to collect more info, but I need some guidance on what is useful and what to ignore. Also, I understand we need some info before/after a swipe, so I can do that as well.

jnuahs, 2015-05-14:
I'm not sure if this if helpful, but I scanned the card with an Android App Mifare Desfire Tool and it lists 4 applications, and each has a number of access keys and a memory address associated with it. It may be that read access is encrypted as well, which might put us SOL on this one.

karakigr, 2018-05-10:
Hello,

I am also interested in NFC Programming and I need to be able using an NFC Reader (like the one utilized in Transportation Buses) to read an IstanbulKart and to charge it also (write data to the card).

Do you have any information regarding the specific communication protocol? Or do you have any idea who could give such an information?

Thanks in advance.

cihanbegburs, 2018-05-10:
Hello,

As far as i know, Mifare desfire protocol is used for IstanbulKart. But you
need to know that it is impossible to write without keys. In additional you
can examine what SAM(Secure access module) is.

karakigr, 2018-05-11:
Ok, thanks for the reply.
I know that Mifare desire protocol is used. I have already used it in another application have implemented. My question is not for the protocol itself but for the data format stored in the card, which application is used for charging, what data need to read to check the balance, etc.

Of Course, I know the AES is used for the encryption of the data and I need to know the key for reading/writing.
So my questions are:
Who is responsible for these keys? For example, the existed retail machines in the buses that are charging the Istanbulkart  they are aware of these keys. If I have a new retail machine, how do I know these keys? Moreover, how to know the data format for charging and displaying the balance of the card?

 
Thanks in advance for your replies!

phcoder, 2018-10-02:
IstanbulKart is locked for everything except serial number. I added reading of serial number to Metrodroid. To be able to read more you need responses to authntication challenge. Official app IstanbulKart queries server for this. Trouble is that server requires a valid Turkish ID with matching name and birthdate. So I couldn't play with this at all as I'm not Turkish.

alpdoga, 2018-10-03:
Thank you for your response.