codebutler/farebot #104: Support for Compass Card (Metro Vancouver, British Columbia, Canada)

daniboy, 2015-05-19:
Compass cards are finally getting into their stable beta phase.

[image]

I've scanned the card and got the following:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<card type="2" id="0123456789abcd" scanned_at="1432055844418">
  <applications>
    <application id="65536">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
  </applications>
  <manufacturing-data>
    <hw-vendor-id>4</hw-vendor-id>
    <hw-type>1</hw-type>
    <hw-sub-type>1</hw-sub-type>
    <hw-major-version>1</hw-major-version>
    <hw-minor-version>0</hw-minor-version>
    <hw-storage-size>24</hw-storage-size>
    <hw-protocol>5</hw-protocol>
    <sw-vendor-id>4</sw-vendor-id>
    <sw-type>1</sw-type>
    <sw-sub-type>1</sw-sub-type>
    <sw-major-version>1</sw-major-version>
    <sw-minor-version>4</sw-minor-version>
    <sw-storage-size>24</sw-storage-size>
    <sw-protocol>5</sw-protocol>
    <uid>-1234567890</uid>
    <batch-no>123456789</batch-no>
    <week-prod>8</week-prod>
    <year-prod>18</year-prod>
  </manufacturing-data>
</card>
```

(anonymized some numeric fields that I imagine are personally identifying)

The card is currently empty (I just got it), I loaded a product on it via their website (I'm a student and eligible for a U-Pass, a monthly pass for post-secondary students) which will be usable starting June 1st. My guess is that the product will be loaded onto the card itself once I tap it in the bus. After I do that I'll post an updated exported XML.

skylogic004, 2015-05-22:
<?xml version="1.0" encoding="UTF-8"?>
<card type="2" id="00000aaaa00000" scanned_at="1432314961824">
    <applications>
        <application id="65536">
            <files>
                <file id="0">
                    <error>java.lang.Exception: Unknown status code: ae</error>
                </file>
            </files>
        </application>
    </applications>
    <manufacturing-data>
        <hw-vendor-id>4</hw-vendor-id>
        <hw-type>1</hw-type>
        <hw-sub-type>1</hw-sub-type>
        <hw-major-version>1</hw-major-version>
        <hw-minor-version>0</hw-minor-version>
        <hw-storage-size>24</hw-storage-size>
        <hw-protocol>5</hw-protocol>
        <sw-vendor-id>4</sw-vendor-id>
        <sw-type>1</sw-type>
        <sw-sub-type>1</sw-sub-type>
        <sw-major-version>1</sw-major-version>
        <sw-minor-version>4</sw-minor-version>
        <sw-storage-size>24</sw-storage-size>
        <sw-protocol>5</sw-protocol>
        <uid>-1111111111</uid>
        <batch-no>222222222</batch-no>
        <week-prod>8</week-prod>
        <year-prod>18</year-prod>
    </manufacturing-data>
</card>

daniboy, 2015-05-28:
I scanned it again at a train station just to see if something would update on the card (my monthly pass doesn't start until June 1st) - the scanner said "insufficient funds: *card updated*", but nothing changed in the exported XML when I scanned it again.
Will report again when I get to use my U-Pass for real :)

Gigahawk, 2015-09-09:
Borrowed a friend's card before and after it was scanned, the only thing that seems to have changed is the "scanned_at=" field, which I'm assuming only encodes the location and perhaps time the card was last scanned. 

Is the Farebot app ignoring some of the data or is the Compass card just more locked down than other fare cards?

daniboy, 2015-09-09:
Looks like it uses a higher encryption :(

Gigahawk, 2015-09-09:
According to Translink, they do have plans on releasing their own app that would serve this purpose sometime in the future, though given how they've been managing Compass, "sometime in the future" probably means three years and a few million dollars from now.

micolous, 2016-01-27:
It looks like that this is a card with no open sectors.  "Unknown status code: ae" is the error you get when you try to read from a DESFire (type=2) card when you don't have the proper keys.  A DESFire card is a fair bit harder to crack than Mifare Classic.

If you peek at this in something like [TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en) it should show you what keys are used for what sectors.  If there's a sector that you can see with that which you can't with Farebot, then it's a bug.  It would be helpful to get some info from that app too.

Given there's only one sector, Compass is probably storing everything in there, and won't have anything readable for you.  Given the application number is 65536 (2^16) this smells like a really lazy implementation of a card structure... probably will change on "production" cards.

Cubic (the contractor who built the Compass system) have built a lot of different systems around the world, so it seems a really weird mistake for them to make.  They have also built systems where there are open sectors on the card for apps to use (like Sydney's Opal card), so it's something that may come in later.  They at least have a history of doing so, so it's more like whether the local agency can/want to wrangle it out of them.

The `scanned_at` is a UNIX timestamp of when you scanned the card on your phone.  It is a reference used for Farebot in the "scanned cards" view.

Gigahawk, 2016-01-27:
Don't have access to my friend's preproduction card, but I do have a scan of one of the first few of the post-production cards before they became available everywhere (if that makes a difference)

Here's the output from TagInfo:

```
** TagInfo scan (version 4.11.59 [β4011059]) 2016-01-27 15:22:31 **

-- INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
MIFARE DESFire EV1 (MF3ICD41)

# DESFire Applications:
1 unknown application

-- NDEF ------------------------------

# NFC data set storage not present:
Maximum NDEF storage size after format: 4094 bytes

-- EXTRA ------------------------------

# Memory information:
Size: 4 kB
Available: 4.0 kB

# IC detailed information:
Capacitance: 17 pF

# Version information:
Vendor ID: NXP
Hardware info:
* Type/subtype: 0x01/0x01
* Version: 1.0
* Storage size: 4096 bytes
* Protocol: ISO/IEC 14443-2 and -3
Software info:
* Type/subtype: 0x01/0x01
* Version: 1.4
* Storage size: 4096 bytes
* Protocol: ISO/IEC 14443-3 and -4
Batch no: 0xBA3494CBA0
Production date: week 25, 2012

-- TECH ------------------------------

# Technologies supported:
ISO/IEC 7816-4 compatible
Native DESFire APDU framing
ISO/IEC 14443-4 (Type A) compatible
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.IsoDep, android.nfc.tech.NfcA, android.nfc.tech.NdefFormatable]
android.nfc.tech.NdefFormatable
android.nfc.tech.IsoDep
* Maximum transceive length: 261 bytes
* Default maximum transceive time-out: 618 ms
* Extended length APDUs not supported
android.nfc.tech.NfcA
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 618 ms

# Detailed protocol information:
ID: 04:95:45:0A:FB:28:80
ATQA: 0x4403
SAK: 0x20
ATS: 0x06757781028000
* Max. accepted frame size: 64 bytes (FSCI: 5)
* Supported receive rates:
	- 106, 212, 424, 848 kbit/s (DR: 1, 2, 4, 8)
* Supported send rates:
	- 106, 212, 424, 848 kbit/s (DS: 1, 2, 4, 8)
* Different send and receive rates supported
* SFGT: 604.1 us  (SFGI: 1)
* FWT: 77.33 ms  (FWI: 8)
* NAD not supported
* CID supported
* Historical bytes: 0x80 |.|

# Memory content:
PICC level (Application ID 0x000000)
* PICC key configuration:
  - AES key
  - PICC key changeable
  - PICC key required for:
    ~ directory list access: no
    ~ create/delete applications: yes
  - Configuration changeable
  - PICC key version: 129

Application ID 0x000001
* Key configuration:
  - 14 (3)DES keys
  - Master key changeable
  - Master key required for:
    ~ directory list access: no
    ~ create/delete files: yes
  - Configuration changeable
  - Key itself required for changing a key
* 1 file present

  - File ID 0x00: Standard data, 384 bytes
    ~ Communication: with MAC
    ~ Read key: key #1
    ~ Write key: key #2
    ~ Read/Write key: key #2
    ~ Change key: master key
    ~ (No access)

--------------------------------------
```

XML output of TagInfo can be found at http://pastebin.com/JWcH85q0
Here's the output from farebot, although I don't think anything has changed:

```
<?xml version="1.0" encoding="UTF-8"?><card type="2" id="0495450afb2880" scanned_at="1453937133074">
  <applications>
    <application id="65536">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
  </applications>
  <manufacturing-data>
    <hw-vendor-id>4</hw-vendor-id>
    <hw-type>1</hw-type>
    <hw-sub-type>1</hw-sub-type>
    <hw-major-version>1</hw-major-version>
    <hw-minor-version>0</hw-minor-version>
    <hw-storage-size>24</hw-storage-size>
    <hw-protocol>5</hw-protocol>
    <sw-vendor-id>4</sw-vendor-id>
    <sw-type>1</sw-type>
    <sw-sub-type>1</sw-sub-type>
    <sw-major-version>1</sw-major-version>
    <sw-minor-version>4</sw-minor-version>
    <sw-storage-size>24</sw-storage-size>
    <sw-protocol>5</sw-protocol>
    <uid>184532421</uid>
    <batch-no>882166874</batch-no>
    <week-prod>37</week-prod>
    <year-prod>18</year-prod>
  </manufacturing-data>
</card>
```

BrainiacV, 2017-05-13:
wow after 2 years, still no support coming from Compass / Translink to help us take advantage of NFC on our devices :(

VMelandri, 2017-06-29:
LOL like if that is happening anytime soon :) Maybe by two-thousand-when-I-retire.

codebutler, 2017-07-02:
Unfortunately there's not much we can do here because the card requires authentication.

closed 2017-07-02 by codebutler