codebutler/farebot #112: Trip list recycles table rows improperly, causing values to move around

BHSPitMonkey, 2015-12-04:
Right now, when I scan my Clipper card and go to the Trips tab, I see several rows that say "Muni" and "Fare". (It literally shows the word "Fare"; there is no actual fare amount for these rides because they are covered by a monthly pass.)

If I scroll down a bit, I can find some BART rides which say "BART", the start/end locations, and "$8.65".

But then, if I scroll back **up**, two of the earlier Muni rows that get revealed show "$8.65" where they said "Fare" previously. (These are obviously not real Muni fare amounts). If I scroll up and down the list some more, the "$8.65" strings move around haphazardly.

This is almost certainly related to recycling table row views while scrolling. My guess is that trips with no fare (like my monthly-pass-covered Muni trips) end up not updating whatever value is present in the recycled row view, and they *should* always be setting it to some value (an empty string, "Fare", "No Fare", "$0.00", etc.).

micolous, 2015-12-28:
I have fixed this bug in my fork:

https://github.com/micolous/farebot/commit/e8beb8eb748f6dada67e71e58c4da82064ceccd9#diff-6e127a842c5092e6205ac5fc5f95c123

codebutler, 2016-05-31:
Thanks, this was fixed by e20bc3d4331abadf3843dea9244fd22d02e3fe9b.

closed 2016-05-31 by codebutler