codebutler/farebot #115: Support for Indonesian KMT Card (Jakarta Commuter Pass)

emirsyehsar, 2016-02-08:
I read that this project could support any transportation card that implements FeliCa or Mifare Classic. So recently KCJ, the operator of commuter rail service in Jakarta began to implement FeliCa on their electronic ticket, the KMT. I saw some FeliCa transportation is already supported by farebot, so I would like to see support for the KMT. Maybe anyone could tell where should I start to read the data inside the FeliCa card?

phcoder, 2018-08-20:
This was added by sybond in Metrodroid

sybond, 2018-08-21:
Hi @emirsyehsar you can try Metrodroid (with KMT support) [here](https://github.com/sybond/farebot/releases/tag/v2.9.34-kmt-debug).

emirsyehsar, 2018-08-21:
@sybond Thank you. I think I might have some addition to the code since I am also trying to "decode" the raw data for the past two months. I think I can move some of my code piece to Metrodroid instead.