codebutler/farebot #121: Support Kiev metro cards

vzaliva, 2016-07-04:
Kiev metro cards are physically compatible with FareBot but it could not decode their contents.
Here is a link to an article (in Russian) which describes with details revere-engineered card protocol and python code snipped to read the card:

https://habrahabr.ru/post/141213/

phcoder, 2018-08-20:
@vzaliva Could you send me a card? I'll pay for the shipping and the card

vzaliva, 2018-08-20:
If you are in US, I can send you my card. If you are in Europe, it is better to find somebody closer to you to send one.

phcoder, 2018-08-25:
I'm likely to travel to US next month. Which region are you in? Bay Area by any chance?

vzaliva, 2018-08-26:
I am in bay area (Saratoga, CA).

phcoder, 2018-09-24:
I got my hands on 2 cards: pre-2016 and post-2016. Later one are MiFare Plus S operating in SL3 mode without any open sectors. So until vulnerability is found we can't read post-2016 cards. Pre-2016 cards can be implemented but I'm not sure it's worth the time

vzaliva, 2018-09-24:
If newer cards are not readable there is not much point to support old ones.
Besides, as now Kiev metro supports both google and apple pay I personally stopped using cards :)

closed 2018-09-24 by vzaliva