codebutler/farebot #122: MetroTransit (Minnesota) GoTo Card Support

jtquen7, 2016-08-28:
Can someone add support for MetroTransit (MIFare Classic) Cards please?

** TagInfo scan (version 4.20.160) 2016-08-27 20:04:08 **
Report Type: External

-- INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
MIFARE Classic (MF1S50)

-- NDEF ------------------------------

# NFC data set storage not present:
Maximum NDEF storage size after format: 716 bytes

-- EXTRA ------------------------------

# Memory size:
1 kB
* 16 sectors, with 4 blocks per sector
* 64 blocks, with 16 bytes per block

-- TECH ------------------------------

# Technologies supported:
MIFARE Classic compatible
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.MifareClassic, android.nfc.tech.NfcA, android.nfc.tech.NdefFormatable]
android.nfc.tech.NdefFormatable
android.nfc.tech.MifareClassic
android.nfc.tech.NfcA
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 20480 ms

# Detailed protocol information:
ID: 9C:0C:37:01
ATQA: 0x0400
SAK: 0x08

# Memory content:
Sector 0 (0x00)
[00] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[01] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[02] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[03] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 1 (0x01)
[04] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[05] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[06] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[07] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 2 (0x02)
[08] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[09] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[0A] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[0B] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 3 (0x03)
[0C] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[0D] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[0E] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[0F] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 4 (0x04)
[10] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[11] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[12] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[13] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 5 (0x05)
[14] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[15] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[16] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[17] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 6 (0x06)
[18] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[19] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[1A] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[1B] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 7 (0x07)
[1C] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[1D] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[1E] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[1F] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 8 (0x08)
[20] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[21] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[22] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[23] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 9 (0x09)
[24] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[25] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[26] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[27] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 10 (0x0A)
[28] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[29] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[2A] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[2B] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 11 (0x0B)
[2C] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[2D] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[2E] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[2F] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 12 (0x0C)
[30] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[31] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[32] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[33] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 13 (0x0D)
[34] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[35] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[36] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[37] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 14 (0x0E)
[38] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[39] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[3A] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[3B] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 15 (0x0F)
[3C] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[3D] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[3E] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[3F] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

r/R=read, w/W=write, i/I=increment,
d=decr/transfer/restore, x=r+w, X=R+W
data block: r/w/i/d:key A|B, R/W/I:key B only,
  I/i implies d, *=value block
trailer (order: key A, AC, key B): r/w:key A,
  W:key B, R:key A|B, (r)=readable key
AC: W implies R+r, R implies r

--------------------------------------

micolous, 2016-09-12:
You'll need to crack your card first, you've got a fully locked Mifare Classic card.

jtquen7, 2016-09-12:
Thanks

-
Sent from Moto X Developer Edition

via CloudMagic Email<https://cloudmagic.com/k/d/mailapp?ct=pa&cv=8.6.38&pv=5.1&source=email_footer_2>

codebutler, 2026-02-15:
Supported in KMP rewrite via `transit/msp-goto/` module, ported from Metrodroid.