codebutler/farebot #134: Support for Bangalore's "Namma Metro" Varshik Card

the-solipsist, 2017-01-27:
Thanks for this app, which seems quite useful.

I wonder if it can be extended to the Bangalore metro card ("[Varshik](http://smartcard.bmrc.co.in/)").  It is either a MIFARE DESFire EV1 or MIFARE DESFire card.  (NXP TagInfo says it is a DESFire card - MF31CD40, while press reports state that it is a EV1 card).  What information is needed to figure out if it will work?

Here is the output I get after scanning it in Farebot.

```
<?xml version="1.0" encoding="UTF-8"?><card type="2" id="04455c4adc4880" scanned_at="1485511978401">
  <applications>
    <application id="4411213">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="1">
          <error>java.lang.UnsupportedOperationException: Value files not yet supported</error>
        </file>
        <file id="2">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="3">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="4">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="5">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="6">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="8">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="9">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="10">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="11">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="12">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
    <application id="5067858">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="1">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="2">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="3">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="4">
          <error>java.lang.UnsupportedOperationException: Value files not yet supported</error>
        </file>
        <file id="6">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="7">
          <error>java.lang.UnsupportedOperationException: Value files not yet supported</error>
        </file>
        <file id="8">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="9">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
    <application id="5264467">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="1">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="2">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="6">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="7">
          <error>java.lang.UnsupportedOperationException: Value files not yet supported</error>
        </file>
        <file id="8">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="9">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
    <application id="5395532">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="1">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="2">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="6">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="7">
          <error>java.lang.UnsupportedOperationException: Value files not yet supported</error>
        </file>
        <file id="8">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
    <application id="5128531">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="1">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="2">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="6">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="7">
          <error>java.lang.UnsupportedOperationException: Value files not yet supported</error>
        </file>
        <file id="8">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
    <application id="5262163">
      <files>
        <file id="0">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="1">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="2">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="5">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="6">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
        <file id="7">
          <error>java.lang.UnsupportedOperationException: Value files not yet supported</error>
        </file>
        <file id="8">
          <error>java.lang.Exception: Unknown status code: ae</error>
        </file>
      </files>
    </application>
  </applications>
  <manufacturing-data>
    <hw-vendor-id>4</hw-vendor-id>
    <hw-type>1</hw-type>
    <hw-sub-type>1</hw-sub-type>
    <hw-major-version>1</hw-major-version>
    <hw-minor-version>0</hw-minor-version>
    <hw-storage-size>24</hw-storage-size>
    <hw-protocol>5</hw-protocol>
    <sw-vendor-id>4</sw-vendor-id>
    <sw-type>1</sw-type>
    <sw-sub-type>1</sw-sub-type>
    <sw-major-version>1</sw-major-version>
    <sw-minor-version>4</sw-minor-version>
    <sw-storage-size>24</sw-storage-size>
    <sw-protocol>5</sw-protocol>
    <uid>1256230364</uid>
    <batch-no>1691846506</batch-no>
    <week-prod>55</week-prod>
    <year-prod>21</year-prod>
  </manufacturing-data>
</card>
```

phcoder, 2018-08-20:
Hey. I think I will need physical instance of the card? Can you send me one? I'll pay for the card and shipping

the-solipsist, 2018-08-26:
Sure thing. Do drop me an email with the address: the.solipsist at gmail