codebutler/farebot #141: Support for Touch n Go cards (Malaysia) (Above Singapore :)

DexterKwok, 2017-06-09:
It would be awesome to support fare checker for Malaysia's transit card Touch n Go. I've tried it and it doesn't work. An encryption keys is needed. I'm willing to help to detect any card value differences.

phcoder, 2018-09-14:
Can you dump it and share several annotated dumps? You might have to acquire keys for the card

sybond, 2018-09-24:
Last week I go to Kuala Lumpur, Malaysia. And managed to have Touch n' Go card. And yeah, we need to acquire the key to able to read the data...
```
** TagInfo scan (version 4.23) 2018-09-21 23:27:40 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
MIFARE Classic (EV1) (MF1S70)

-- NDEF ------------------------------

# No NDEF data storage present:
Maximum NDEF storage size after format: 3356 bytes

-- EXTRA ------------------------------

# Memory size:
4 kB
* 32 sectors of 4 blocks and 8 sectors of 16 blocks
* 256 blocks, with 16 bytes per block

-- FULL SCAN ------------------------------

# Technologies supported:
MIFARE Classic compatible
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareClassic, android.nfc.tech.NdefFormatable]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 618 ms

# Detailed protocol information:
ID: 2F:87:1B:F4
* Non-unique ID
ATQA: 0x0200
SAK: 0x18
ATS: 0x0E75778002D00700020090E3F1EE0000
* Max. accepted frame size: 64 bytes (FSCI: 5)
* Supported receive rates:
- 106, 212, 424, 848 kbit/s (DR: 1, 2, 4, 8)
* Supported send rates:
- 106, 212, 424, 848 kbit/s (DS: 1, 2, 4, 8)
* Different send and receive rates supported
* SFGT: 302.0 us  (SFGI: 0)
* FWT: 77.33 ms  (FWI: 8)
* NAD not supported
* CID supported
* Historical bytes: 0xD00700020090E3F1EE |.␇␀␂␀....|

# Memory content:
Sector 0 (0x00)
[00] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[01] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[02] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[03] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

Sector 1 (0x01)
[04] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[05] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[06] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[07] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)
[... 346 lines cut ...]
[F1] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F2] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F3] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F4] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F5] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F6] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F7] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F8] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[F9] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[FA] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[FB] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[FC] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[FD] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[FE] ???  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[FF] ???  XX:XX:XX:XX:XX:XX --:--:-- -- XX:XX:XX:XX:XX:XX
          (unknown key)                 (unknown key)

r/R=read, w/W=write, i/I=increment,
d=decr/transfer/restore, x=r+w, X=R+W
data block: r/w/i/d:key A|B, R/W/I:key B only,
  I/i implies d, *=value block
trailer (order: key A, AC, key B): r/w:key A,
  W:key B, R:key A|B, (r)=readable key
AC: W implies R+r, R implies r
```

phcoder, 2018-09-24:
Did you try the MiFare Classic Tool with keyset from github https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/assets/key-files/extended-std.keys ?

sybond, 2018-09-24:
@phcoder no luck using MCT latest keys

sumanblack666, 2019-04-03:
@phcoder i got few dumps if u want

phcoder, 2019-05-05:
@sumanblack666 Can you send them to me privately?

sumanblack666, 2019-05-05:
Hi drop me an email to sumanblack666@gmail.com . I'll reply with few dumps.

phcoder, 2019-07-02:
This is now committed and released on Metrodroid

DanisyEisyraf, 2023-11-08:
@sumanblack666 do you still have the dumps?

DanisyEisyraf, 2023-11-08:
@DexterKwok Its been a while, did you managed to retrieve any keys?

sumanblack666, 2023-12-16:
> @sumanblack666 do you still have the dumps?

Not sure whether I have it in my old hdd. I've to check n see. Btw can I know why u need the dump?