codebutler/farebot #149: CEPAS 2.0 card 'Invalid Response'

Vantablack, 2017-08-23:
## Issue
Hi I'm having an issue when attempting to scan a CEPAS 2.0 card and I'm getting **Invalid Response**

[image]

## Error 1
After some digging, I got an error log which is [full log here](https://gist.github.com/Vantablack/546383d301d0f96b12ace544253593e8#file-error1-log)

```
E/com.codebutler.farebot.app.feature.main.MainActivity: Invalid response java.lang.Exception: Invalid response at com.codebutler.farebot.card.desfire.DesfireProtocol.sendRequest(DesfireProtocol.java:141)
```

From here I deduced that **DesfireTagReader** is called instead of **CEPASTagReader**. So I swapped the order in `TagReaderFactory.kt`

## Error 2
Error 1 was fixed but another was created (Error 2), [full log here](https://gist.github.com/Vantablack/546383d301d0f96b12ace544253593e8#file-error2-log)

```
E/AndroidRuntime: FATAL EXCEPTION: main Process: com.codebutler.farebot, PID: 27383 io.reactivex.exceptions.OnErrorNotImplementedException: Invalid index 3, size is 0
```

And this was caused by the ArrayList having a size() of 0 and so I tried using Arrays [] instead by [following what was done in v0.5](https://github.com/codebutler/farebot/blob/v0.5/src/com/codebutler/farebot/cepas/CEPASCard.java#L47)

[image]

## Error 3

Error 2 was fixed but Error 3 was created, [full log here](https://gist.github.com/Vantablack/546383d301d0f96b12ace544253593e8#file-error3-log)

It was due to invalid purses having null values for `getCAN`, `getCSN`, `getLastCreditTransactionHeader`, `getLastTransactionRecord` and `getIssuerSpecificData` in `CEPASPurse.java`, and adding @Nullable seems to solve the issue.

Similar to [this merge request](https://github.com/codebutler/farebot/pull/3), its seems like data is stored in purse 3 (I've tested 3 recent cards).

## Possible Fix

I'm not sure if I should open a merge request but here's my current fixes.

https://github.com/Vantablack/farebot/commit/a0c1f97f433c2d363203936b1384c31c12b0be06

[image]

DexterKwok, 2017-08-30:
yup, I am having trouble scanning my Nets Card, "Invalid Response"

codebutler, 2017-09-01:
Please do open a pull request, thank you!

closed 2017-09-04 by codebutler

Vantablack, 2017-09-05:
Oh hey @codebutler! Thanks for closing this issue!

themakerman, 2017-10-24:
@Vantablack Are you able to read the card now after the fix?

Vantablack, 2017-10-25:
@themakerman I currently do not have access to an Android device so can’t check for you. However it did work after I applied some changes see (https://github.com/Vantablack/farebot/commit/a0c1f97f433c2d363203936b1384c31c12b0be06) which I believe was committed in https://github.com/codebutler/farebot/commit/d3637bdff63201c2c65d13167c2cd7532f330651

themakerman, 2017-10-30:
@Vantablack it did worked. It's not updated in the production though on playstore.

themakerman, 2017-10-30:
@Vantablack Do you think since IOS 11 NFC has came up we can use the same cepas protocol for iOS users across island? I think majority people use iPhone in SG. Have you made any research on iOS capabilities to read SG transit cards?

Vantablack, 2017-10-31:
@themakerman I don’t think iOS 11 Core NFC supports sending custom APDU. From the [documentation](https://developer.apple.com/documentation/corenfc), I think it only supports reading NDEF formatted tags only?

See [iOS11 Core NFC and ISO 14443](https://stackoverflow.com/questions/44424301/ios11-core-nfc-and-iso-14443)

commit d3637bdf (2017-09-04): Fix for CEPAS card

Fix for 'Invalid Response' when scanning a CEPAS card

Fixes #149