codebutler/farebot #151: Null reference error reading PASMO

dhleong, 2017-09-21:
```
net.kazzz.felica.lib.FeliCaLib$IDm.getBytes()
```

On latest version of Android using the Pixel. Repro is 100%: just scan the pasmo.

For what it's worth, the card is old (~7 years) and used to have a commuter pass attached, though I believe it used to work with an older version of this app. Other suica reader apps don't have this problem.

Please let me know if there's anything else I can provide to help!

codebutler, 2017-11-17:
I just published a new build to the beta channel that will send these errors to the crash reporting service.  Please follow these steps:

1. Please opt-in using this link: https://play.google.com/apps/testing/com.codebutler.farebot
2. Install the 3.0.1 update through the Play Store on your device
3. Reproduce the crash a few times.

Let me know when this is done and I'll check Crashlytics for more detail and can hopefully get this resolved soon. Thank you!

dhleong, 2017-11-29:
Sorry for the delay. I just tried scanning and it worked no problem. For what it's worth, I got back from Japan a couple weeks ago, so the card has changed since my first attempts. The most recent transaction when it crashed was from December 2011.

codebutler, 2017-11-29:
Thanks for the update, I'm going to close this for now.  Please re-open if anyone can still reproduce and has time to help me debug or could mail me a non-working card.

closed 2017-11-29 by codebutler

mranostay, 2019-04-20:
Noticed I get this message when I don't have the card directly over the NFC reader on my Pixle. But if I do it works fine.

Scrxtchy, 2019-09-09:
I've had a similar situation to @mranostay, direct contact with the device prevented an error on the current version, as this bug would have been fixed years ago