codebutler/farebot #153: Add support for URBS bus cards (Curitiba, BR)

calchuchesta, 2018-01-08:


phcoder, 2018-08-20:
Hey, can you supply annotated dumps?

calchuchesta, 2018-08-24:
...of what, exactly?

phcoder, 2018-08-25:
Of cards in question

codebutler, 2026-02-15:
This issue is being closed due to age (opened in 2018). 

FareBot has undergone significant development since this was originally filed, including a complete rewrite as a Kotlin Multiplatform app with extensive backports from [Metrodroid](https://github.com/metrodroid/metrodroid).

If you still need support for this card system / feature:
- Please open a new issue with updated information
- If possible, include sample card dumps from recent cards
- Check if Metrodroid has added support for this system (we can backport it)

Thank you for your interest in FareBot!