codebutler/farebot #154: Add support for Ventra tickets (Chicago, IL)

bmintz, 2018-02-09:
The CTA Ventra cards present as MasterCard NFC credit cards, and so balance information cannot be read from them. However, they also sell single use [tickets and passes](http://www.transitchicago.com/fares/) that last multiple days. These are Mifare Ultralight C cards. I can provide hex dumps of cards in various states (used, unused, one transfer left, etc) if necessary.

phcoder, 2018-08-09:
I would be interested in the dumps. Obviously no promises.

bmintz, 2018-08-09:
https://ptpb.pw/Ammh.tar.xz

phcoder, 2018-08-09:
What is the number printed on them? They probably have some kind of serial

phcoder, 2018-08-09:
Also any fields you'd like to be read, we need their value in those dumps

bmintz, 2018-08-10:
@phcoder i don't have that info any more. all information i have left is in the ticket, except for one ticket.

phcoder, 2018-08-10:
@bmintz, even having a single ticket with matching serial is helpful. Generally the more data you collect (time of validation, serial, place of validation and so on), better are the chances of support.

phcoder, 2018-08-20:
Turns out that they're the same format as Compass tickets from Vancouver. Try scanning them with Metrodroid