codebutler/farebot #180: iOS 13 Support?

mrarfarf, 2019-06-22:
iOS 13 expands it's NFC Api's to (from what I can tell) allow all the functions needed (FINALLY) to allow for a port of farebot.

Any chance this is in the timeline?

phcoder, 2019-07-02:
@mrarfarf I prepared Metrodroid to be mostly in multiplatform-compatible way. So this shouldn't be too difficult now. I'd be interested in learning How to write missing parts but have no hardware for it.

mrarfarf, 2019-07-02:
What hardware do you need?

Best Regards,
Christopher

phcoder, 2019-07-02:
I looked at the API a bit and it looks like it doesn't support Mifare Classic. But Desfire, Felica, Calypso, Tmoney and NfcV should work. China and Cepas srandards are a gamble. I think Desfire could be a good first step.

mrarfarf, 2019-07-02:
Nice. So so with seafarers support, it could be possible to support orca cards? What would you need hardware wise to get this going? I’d be willing to help.

Best Regards,
Christopher

phcoder, 2019-07-02:
I need some way to run SDK. I wonder if I can get some cloud-based Mac for this. Or get some cheap mac desktop. I'll have to check what are system requirements. I will later need an iPhone to test with. It looks like lowest with NFC and iOS 13 compatiblity is IPhone 6.

phcoder, 2019-07-02:
Orca should be compatible looking from quick look atdocumentation

mrarfarf, 2019-07-02:
God I hate iOS spellcheck. Seafarer = desfare. Yeah, I’m looking for an app to clone my care so I don’t have to carry it around, I think that’s what your app does? I’ve never used it since I have iOS and it’s never had the apis until iOS 13. 

Anyway, if I can help at all, please let me know!

Best Regards,
Christopher

mrarfarf, 2019-07-02:
Hmm.. ok, what country are you located in? Maybe a Mac vpc via aws? 

Best Regards,
Christopher

phcoder, 2019-07-03:
No. Neither Metrodroid, nor Farebot do any cloning or emulating. In case of desfire it's not practically possible to clone or emulate it without partnership with card issuer (in this case ORCA). The only thing Farebot/Metrodroid does is showing the contents of the card in human-readable format

codebutler, 2026-02-14:
KMP rewrite supports modern iOS natively.