codebutler/farebot #185: FareBot app crashes when attempting to delete the card history in the app menu

smkent, 2019-10-07:
I have FareBot on a Google Pixel 3, and I use it to check the value on my ORCA card. I can scan the card without issue.

After reading a card, attempting to delete the card history using the menu option in the FareBot app doesn't work and causes the app to crash. To delete the data, I have to clear FareBot's app data in Android settings.

ayshen, 2022-05-25:
```
java.lang.IllegalStateException: Cannot access database on the main thread since it may potentially lock the UI for a long period of time.
 	at androidx.room.RoomDatabase.assertNotMainThread(RoomDatabase.java:261)
 	at androidx.room.RoomDatabase.beginTransaction(RoomDatabase.java:327)
 	at com.codebutler.farebot.persist.db.dao.SavedCardDao_Impl.delete(SavedCardDao_Impl.java:93)
 	at com.codebutler.farebot.persist.db.DbCardPersister.deleteCard(DbCardPersister.kt:10)
 	at com.codebutler.farebot.app.feature.history.HistoryScreen.onDeleteSelectedItems(HistoryScreen.kt:170)
 	at com.codebutler.farebot.app.feature.history.HistoryScreenView$onAttachedToWindow$1$1.onActionItemClicked(HistoryScreenView.kt:80)
 	at androidx.appcompat.app.AppCompatDelegateImpl$ActionModeCallbackWrapperV9.onActionItemClicked(AppCompatDelegateImpl.java:2171)
 	at androidx.appcompat.view.StandaloneActionMode.onMenuItemSelected(StandaloneActionMode.java:141)
 	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:840)
 	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
 	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:991)
 	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:981)
 	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
 	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
 	at android.view.View.performClick(View.java:7455)
 	at android.view.View.performClickInternal(View.java:7432)
 	at android.view.View.access$3700(View.java:835)
 	at android.view.View$PerformClick.run(View.java:28810)
 	at android.os.Handler.handleCallback(Handler.java:938)
 	at android.os.Handler.dispatchMessage(Handler.java:99)
 	at android.os.Looper.loopOnce(Looper.java:201)
 	at android.os.Looper.loop(Looper.java:288)
 	at android.app.ActivityThread.main(ActivityThread.java:7870)
 	at java.lang.reflect.Method.invoke(Native Method)
 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
```

It looks like the `for ((savedCard in items)` in HistoryScreen.onDeleteSelectedItems should be moved to a background thread and post back to the UI thread to `loadCards()` afterward?

codebutler, 2026-02-14:
Fixed in KMP rewrite. Card deletion uses SQLDelight + coroutines (viewModelScope.launch), no main-thread database access.

closed 2026-02-14 by codebutler